chuckwagoncomputing/interactive-pinout #3: Make columns more configurable

rusefillc, 2023-03-11:
Example image mentions "TS Name" which only makes sense for rusEFI specific usage. As one more step towards total domination it should be possible not to have "TS Name" column and overall I guess make columns more configurable. Wire color is a column which is barely used even in RE.

[image]

chuckwagoncomputing, 2023-03-11:
The columns are specified by the user in the workflow file.
[The rusEFI workflow], for example, passes the following `columns` parameter to interactive-pinout.
```
columns: |
  {
  "pin":"Pin Number",
  "ts_name":"TS Name",
  "type":"Type",
  "function":"Typical Function",
  "color":"Pigtail Color"
  }
```

Does this address the issue, or is your concern the initial perception that these are the only possible columns?

rusefillc, 2023-03-11:
Ohhh it's already configurable, I am a bit behind :)

closed 2023-03-11 by rusefillc