chuckwagoncomputing/interactive-pinout #5: simplify maintenance of pinout docs via  "dynamic views" feature resolved

mi-hol, 2023-03-26:
The main document for pinouts is the YAML definition, but often multiple iterations are required to create a fully correct definition.
At the same time certain subsets of the definition "views" are required in docs to illustrate facts like grouping of all pins used for "power" or "ignition".
Currently  these views have to be created manually via a copy & paste process.
Unfortunately later changes in YAML need to be applied to these "disconnected copies" too.
There is a high risk of failure because original author(s) and maintainer(s) are frequently different persons.

This risk could be avoided by defining a "dynamic view" as a kind of query for the YAML definition that gets resolved into a static markdown text at time of document build/lint.

mi-hol, 2023-03-26:
https://mdxjs.com/docs/extending-mdx/ could be an approach to implement this feature

chuckwagoncomputing, 2023-03-27:
I think I understand now, but this is not within the scope of this tool.
This tool generates interactive pinouts with HTML/CSS/JS, not markdown files.

closed 2023-03-27 by chuckwagoncomputing

mi-hol, 2023-04-02:
HTML can be embedded into 

Would the[ pick operator](https://mikefarah.gitbook.io/yq/operators/pick) be the mechanism to select the required fields?

chuckwagoncomputing, 2023-04-05:
Maybe; try it and find out.