chuckwagoncomputing/interactive-pinout #6: Reduce manual labor: auto-generate C header?!

rusefillc, 2023-03-29:
I know title sounds odd please hear me out.

Problem statement:
I am clearly still typing same thing twice:
first I open yaml and I do
```
  - pin: B23
# H144_OUT_PWM4
    id: C8
    class: outputs
    ts_name: --- VTC VVT
    function: VTC VVT
```

(not how I have to say ID = C8 for my custom java in rusEFI while leaving a human-readable comment H144_OUT_PWM4 which is hellen_meta.h)

and then I go edit boards.cpp file where I type
```
    engineConfiguration->vvtPins[0] = H144_OUT_PWM4;

```

Note how I am manually keeping things in sync?

No idea what exactly I am looking for but for instance

if I have machine-generated
```
#define B23_VTC_VVT C8
```

I would at least 
```
    engineConfiguration->vvtPins[0] = GPIO::B23_VTC_VVT ;
```

which is one step closer to not manually syncing yaml and C

chuckwagoncomputing, 2023-03-29:
Not in scope of this repo, but seems doable. Make an issue in rusefi/rusefi and we'll discuss there.

closed 2023-03-29 by chuckwagoncomputing

rusefillc, 2023-03-29:
https://github.com/rusefi/rusefi/issues/5206