chuckwagoncomputing/interactive-pinout #10: Feature request: externalize image coordinates

rusefillc, 2023-07-17:
I keep reusing same image for multiple different pinouts meaning I copy-paste coordinate pin coordinates for instance https://github.com/rusefi/rusefi/blob/2715ccd55aa859ecf0577d7c30f65568d1cf8e9d/firmware/config/boards/hellen/hellen121vag/connectors/main.yaml#L420 vs https://github.com/rusefi/rusefi/blob/5f527317d6c6a42ea254775803a9ea240d639c21/firmware/config/boards/hellen/hellen121nissan/connectors/main.yaml#L361

Would it be cool to allow reuse of coordinate file while I am reusing image file?

chuckwagoncomputing, 2023-07-17:
My initial yaml schema would have made this easy to do: I had the `pins` section containing the coordinates as a child of the `image` section, but at some point it ended up being a child of the `info` section instead. If I had stuck with the original design, the `image` section could have been put in another file, and we could have a setting like this in the main file:
```
info:
  image:
    import: shared-image.yaml
```
I'm very tempted to go back to the original design, or possibly support both.

chuckwagoncomputing, 2023-07-17:
I've added this and will be releasing it shortly.
Defining:
https://github.com/chuckwagoncomputing/rusefi/tree/7ac8a96681835b0bbe6af9565daf2a828969621e/hardware/connectors/121
Sourcing:
https://github.com/chuckwagoncomputing/rusefi/blob/7ac8a96681835b0bbe6af9565daf2a828969621e/firmware/config/boards/s105/connectors/s105.yaml

chuckwagoncomputing, 2023-07-17:
Released in [2.7](https://github.com/chuckwagoncomputing/interactive-pinout/releases/tag/2.7)

closed 2023-07-17 by chuckwagoncomputing