chuckwagoncomputing/interactive-pinout #9: What do I do wrong?

rusefillc, 2023-07-17:
I have a small repo https://github.com/rusefi/adapters where I would like to set this amazing https://github.com/chuckwagoncomputing/interactive-pinout

job seems to succeed but does not seem to produce anything, why?

```
Processing: ./hyundai-pb-154-superseal-adapter/connectors/60pins.yaml
WARNING: Missing yaml cid field in info section of ./hyundai-pb-154-superseal-adapter/connectors/60pins.yaml
Image: 60pins.jpg
Completed processing 0 mappings
```

rusefillc, 2023-07-17:
Now I am even more confused

```
Processing: ./hyundai-pb-1[54](https://github.com/rusefi/adapters/actions/runs/5571426411/jobs/10176448604#step:4:55)-superseal-adapter/connectors/26pins.yaml
WARNING: Missing yaml cid field in info section of ./hyundai-pb-154-superseal-adapter/connectors/26pins.yaml
Image: 26pins-4k.jpg
Processing: ./hyundai-pb-154-superseal-adapter/connectors/60pins.yaml
WARNING: Missing yaml cid field in info section of ./hyundai-pb-154-superseal-adapter/connectors/60pins.yaml
Image: 60pins.jpg
Completed processing 1 mappings
```

chuckwagoncomputing, 2023-07-17:
Yep I had an off-by-one error.

closed 2023-07-17 by chuckwagoncomputing