chuckwagoncomputing/interactive-pinout #14: I can only click on one pin

rusefillc, 2023-11-07:
pinout definition https://github.com/rusefi/stim/tree/main/digital_inputs/connectors produces https://rusefi.com/docs/pinouts/stim/

clicking on any pin produces table above full pinout as expected

[image]

but clicking any other pin next does not produce similar table for that other pin

[image]

work-around: hit F5 to refresh and click on whatever (one) pin again

chuckwagoncomputing, 2023-11-07:
Looks like something is confusing the minifier. You can work around that with `debug: "true"` in the workflow until I get it figured out.

rusefillc, 2023-11-07:
just confirming that https://github.com/rusefi/stim/commit/96152c581a383cd3b6963ee8217179f67ec73588 work-around has helped

chuckwagoncomputing, 2023-11-08:
https://github.com/chuckwagoncomputing/interactive-pinout/commit/d23e8c0866c16bee779bd1871192f954f0435599 has kinda fixed it but there's still some weird minifying stuff that I think is fixed in https://github.com/tdewolff/minify/releases/tag/v2.20.6

chuckwagoncomputing, 2023-11-11:
The latest release of minify has fixed this issue.

closed 2023-11-11 by chuckwagoncomputing