chuckwagoncomputing/interactive-pinout #16: add validation of "type" field values

mi-hol, 2023-12-16:
There should be a list of allowed values for "type" field in documentation.
Meaning of the type values should be explained there too.

As I understand it, the currently allowed value list and meaningseems to be: 

- at (Analog temperature)
- gnd (Ground)
- sgnd (Sensor Ground)
- pgnd (Power Ground)
- gp_high (General Purpose - high side?)
- gp_low (General Purpose - low side?)
- gp_pp (General Purpose - ?)
- ign (Ignition)
- inj (Injection)
- ls (Low side?) (identical to gp-low?)
- TPS (Throttle Position Sensor)
- vr (variable-reluctance (VR) trigger sensor)

chuckwagoncomputing, 2023-12-16:
The "types" are defined by the project using interactive-pinout, e.g. for rusEFI in https://github.com/rusefi/rusefi/blob/master/.github/workflows/gen-pinouts.yaml.
I agree that documenting these could be useful for the rusEFI project, however, this repo is not the place to do it.

closed 2023-12-16 by chuckwagoncomputing

mi-hol, 2023-12-16:
@chuckwagoncomputing please reopen, because currently "types" the referenced file are really hard to understand. A better reference is needed from my view that should be included or linked in the using repos. The "table" in first comment was the first trial to improve this open topic.