chuckwagoncomputing/interactive-pinout #17: colors: complete type color vs specific wire color

rusefillc, 2023-12-16:
At the moment there is type-based coloring and "color" as just a text column. Asking for more:

I would like to document recommended wire colors. Injector one green wire, injector two white, injector three blue.

No idea how to visually display that :(

https://rusefi.com/docs/pinouts/proteus/?connector=black35&pin=15
https://rusefi.com/docs/pinouts/proteus/?connector=black35&pin=4

chuckwagoncomputing, 2023-12-16:
This is a good idea. Do you propose that only wire colors be shown for pinouts with that option enabled, or would you like to somehow indicate both type and wire color with color?

Also, would it be better to have a per-pinout flag, or just automatically use the color field if any pin in the pinout has it?

rusefillc, 2023-12-16:
Type info is also valuable. I am not good with visuals like @ksmola not sure what's the way to display both type and wire colors

hold on, you use two colors already! what is the current logic - how is inner color not always white? I also assume we do not have control over inner color if text is always black :(

maybe add another outer ring of new color as long as screen resolution is sufficient? and hide optional outer ring on smaller resolutions?

[image]

As for flexibility at the moment you are highly configurable maybe too configurable since it comes as extra dev effort?. maybe just a "useColorColumnForOuterRing" option if anyone wants not to have it?

chuckwagoncomputing, 2023-12-17:
The inner color indicates the selection and other pins which are the same type as the selection.
Adding another ring doesn't seem feasible to me, as the text size is already limited by the amount of available space in some pinouts, e.g. https://rusefi.com/docs/pinouts/hellen/hellen128/

rusefillc, 2023-12-18:
What about wire color instead of type color when mouse over?

chuckwagoncomputing, 2023-12-19:
I don't really like that idea because it's very non-obvious what the color means, and also wire color could be useful for the printed version.
What do you think of having a toggle under the pinout to switch between type colors and wire colors?

rusefillc, 2023-12-20:
> What do you think of having a toggle under the pinout to switch between type colors and wire colors?

I like this idea!

chuckwagoncomputing, 2024-01-19:
I've created a toggle, but there's a small issue - the microRusEFI pinout I have been testing on also has "right" in some of the color fields, so those stay on the color associated with their type.  I'm guessing this is to specify a branch of the harness. This should maybe be moved to a new column.

chuckwagoncomputing, 2024-01-19:
Here you can try it out:
[microrusefi.zip](https://github.com/chuckwagoncomputing/interactive-pinout/files/13982934/microrusefi.zip)

closed 2024-01-19 by chuckwagoncomputing