chuckwagoncomputing/interactive-pinout #18: maybe empty yaml glitch

rusefillc, 2024-01-18:
i have too many yaml files intending to upload all into subfolders and have .htaccess directory listing give me an index of all folders

problem: at the moment attached index.html in the root folder - it has no info in it just JS header

* maybe add an html comment spelling name of source file?
* maybe return error code if zero pins are defined?
* any other ideas how can I find the source file or prevent the glitch?
[index.zip](https://github.com/chuckwagoncomputing/interactive-pinout/files/13979335/index.zip)

chuckwagoncomputing, 2024-01-19:
I have added an warning for that, and the corresponding `warning-no-pins` configuration field.
If you set it to `skip`, it should not generate the index.html for that yaml.
Does this solve this issue?

rusefillc, 2024-01-19:
``warning-no-pins`` or ``error-no-pins``? what's the default value?

my challenge is I've lost track of all interactive-pinout usages so adding new argument... Well just realized that I am screwed any way since I would have to go bump version anyway.

chuckwagoncomputing, 2024-01-19:
https://github.com/rusefi/rusefi/blob/0495c9fd4c5c013690813b8606ef6bf3821a1199/.github/workflows/gen-pinouts.yaml#L23 sets the behavior for warnings: `warnings: "false"`
Individual warning types can override that, e.g. the next line, `warning-dupe: "error"`

I can PR with the warning config and point to a new release (which I haven't made yet)

rusefillc, 2024-01-19:
> I can PR with the warning config and point to a new release (which I haven't made yet)

if you are confident in the change maybe just make a release and i will report if anything break? overall how seriously do you take the whole testing/lifecycle?

chuckwagoncomputing, 2024-01-19:
There's no unit testing or anything like that. I just poke around in several rusEFI pinouts and see if I can spot anything amiss.

closed 2024-01-19 by chuckwagoncomputing