chuckwagoncomputing/interactive-pinout #22: pin line syntax

rusefillc, 2024-10-22:
At the moment we have to manually define coordinate of each pin

for the image as an example I would love to only provide coordinates of 26A and 34A, and have the code interpolate all the coordinates between?

[image]

chuckwagoncomputing, 2024-10-28:
It would be possible to get the numeric portion of the pin number, then replace it with itself  increased and decreased until the next lower and higher pins with coordinates are found. This feels like a edge-case minefield, but I can't think of any that wouldn't be solved with a few extra coordinates, or maybe by adding options to manually supply the next higher and lower pins with coordinates in the row..
```yaml
- pin: K30
  x: 2340
  y: 571
- pin: K31
- pin: K32
- pin: K33
- pin: K34
  x: 2633
  y: 571

rusefillc, 2024-10-28:
@chuckwagoncomputing I agree that this could get pretty confusing pretty fast if I would ask for K1-11 to K1-43 range of pins :(

chuckwagoncomputing, 2024-11-03:
Here's some code that satisfies that case, if you manually give it the bounding pins.
```js
function findBetween(a, b, p) {
        var a_ns = a.pin.match(/\d+/g)
        var b_ns = b.pin.match(/\d+/g)
        var n_ns = p.pin.match(/\d+/g)
        var n = 0;
        for (var i = 0; i < a_ns.length; i++) { 
                if (a_ns[i] != b_ns[i]) { 
                        a.n = a_ns[i];
                        b.n = b_ns[i];
                        n = n_ns[i];
                        break  
                }
        }
        return brange(a, b, n);
}
 
function range(x1, x2, n1, n2, n) {
        n1 = parseInt(n1);
        n2 = parseInt(n2);
        x1 = parseInt(x1);
        x2 = parseInt(x2);
        n = parseInt(n);  
        return x1 + (x2 - x1) * ((n - n1) / (n2 - n1));
}
 
function brange(p1, p2, n) {
        return {
                x: range(p1.x, p2.x, p1.n, p2.n, n),
                y: range(p1.y, p2.y, p1.n, p2.n, n)
        };
}
```
The `match` function gets the numeric substrings from the pin name and puts them in an array, e.g. `"K1-43"` -> `[ "1", "43" ]`, then loops through the array and finds those that are different between the first and last pins, then uses those for the interpolation.
This assumes that all the pin names in a range will have the same number of numeric substrings.

My next task is to find the bounding pins automatically.

chuckwagoncomputing, 2024-11-03:
It would be easy to find the bounds if we could assume the pins were sorted, i.e. just use the last and next pins in array that have coordinates.

chuckwagoncomputing, 2024-11-04:
This feature is now in master. I haven't exhaustively tested it, just with a few:
A-B connector of https://rusefi.chuckwagoncomputing.com/docs/pinouts/hellen/alphax-8chan-rev-A/
https://github.com/chuckwagoncomputing/rusefi/commit/3f0be5fc9d1a3e37129ab0c7e0796c5da23e6297
Black23 connector of https://rusefi.chuckwagoncomputing.com/docs/pinouts/proteus/
https://github.com/chuckwagoncomputing/rusefi/commit/20bcb55b1a8a37252d556e0c8ed583775dca95e3

closed 2024-11-04 by chuckwagoncomputing